STMicroelectronics/stm32g4xx-hal-driver #1: [timer] - ONEPULSEMODE defines wrong documentation

scachemaille, 2020-11-11:
https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/8b4518417706d42eef5c14e56a650005abf478a8/Inc/stm32g4xx_ll_tim.h#L608-L609

I think that the comment on these lines are actually inverted..

ASELSTM, 2020-11-13:
Hi @scachemaille,

Thank you for this report.

Indeed, the comment  are inverted and they should rather be:

```diff
-*#define  LL_TIM_ONEPULSEMODE_SINGLE             TIM_CR1_OPM         /*!< Counter is not stopped at update event */ 
+*#define LL_TIM_ONEPULSEMODE_SINGLE             TIM_CR1_OPM         /*!< Counter stops counting at the next update event*/     
-*#define  LL_TIM_ONEPULSEMODE_REPETITIVE        0x00000000U          /*!< Counter stops counting at the next update event*/ 
+*#define LL_TIM_ONEPULSEMODE_REPETITIVE        0x00000000U          /*!< Counter is not stopped at update event */
```

A bug tracker has been opened internally and the fix will be published in the frame of a future publication. We cannot share a date for the moment. Stay tuned and thank you once more for your contribution and for your patience.

With regards,

ASELSTM, 2020-11-13:
ST Internal Reference: 96303

ASELSTM, 2021-03-04:
Hi @scachemaille,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.2.1 of the stm32g4xx_hal_driver published recently on GitHub.

Thank you again for having reported.

With regards,

closed 2021-03-04 by ASELSTM