STMicroelectronics/stm32g4xx-hal-driver #4: HAL_FDCAN_GetRxMessage doesn't check for buffer size

Chris-Idema-WizzDev, 2022-01-18:
https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/19bed7c9f84aa74aba98c6d1ea54bed39bd8e61a/Inc/stm32g4xx_hal_fdcan.h#L1228

HAL_FDCAN_GetRxMessage doesn't check for buffer size. Since there is no function in this header file to request the size of the message before reading you are forced to allocate an array of 64 bytes to be sure. A bounds check would prevent buffer overflow and reading the size of the message before copying it would be helpful too,

ASELSTM, 2022-02-09:
ST Internal Reference: 122463

RJMSTM, 2024-02-14:
Hello,

I hope you are fine. The issue you reported has been fixed in the frame of version v1.2.3 of the STM32CubeG4 published recently on GitHub. Thank you again for having reported.

Thank you again for your contribution.
With regards,

closed 2024-02-14 by RJMSTM