STMicroelectronics/stm32g4xx-hal-driver #5: Struct member DataLength name and discription misleading

Chris-Idema-WizzDev, 2022-01-18:
https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/19bed7c9f84aa74aba98c6d1ea54bed39bd8e61a/Inc/stm32g4xx_hal_fdcan.h#L212

The struct member DataLength (found in multiple structs) has a misleading name and description.
It is neither data length in bytes nor a CAN Data Length Code (DLC). It is a shifted DLC value. No conversion function/macro is provided (DLCtoBytes is static).

ASELSTM, 2022-02-08:
ST Internal Reference: 122438

ALABSTM, 2024-01-04:
Hi @Chris-Idema-WizzDev,

Best wishes for the new year 2024. Thank you again for your report. The following fix has been implemented in version 1.2.3 which has been released a couple of weeks ago:

* The definitions of the `FDCAN_DLC_BYTES_X` macros are now shift-free so they comply to the definition of the DLC.

https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/1e886abc2c0005b402941edde93a77530861a532/Inc/stm32g4xx_hal_fdcan.h#L610-L625

* The shifts have been moved to the code as in the example shown below.

https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/1e886abc2c0005b402941edde93a77530861a532/Src/stm32g4xx_hal_fdcan.c#L2318-L2319

With regards,

closed 2024-01-04 by ALABSTM