STMicroelectronics/stm32g4xx-hal-driver #8: I2C Hal driver in DMA mode is broken

olliw42, 2023-11-27:
**Describe the set-up**
 * STM32G491RE custom board
 * STM32CubeIDE Version: 1.13.2 Build: 18220_20230914_1601 (UTC)

**Describe the bug**
When using the I2C hal driver with DMA the communication to an OLED display doesn't work. When using the I2C driver in IT mode the display works fine. The project firmware is found here https://github.com/olliw42/mLRS and the board schematics here https://github.com/olliw42/mLRS-hardware/tree/master/olliw-stm32-based/tx-module-02, if this should be of any help.

Tests have confirmed that the issue is specifically related to the I2C hal driver, that is, is located in the file stm32g4xx_hal_i2c.c

The board including OLED was working fine since ca 1.5 years, up to using commit @75ce54b. The issue appeared recently with upgrading to commit @74ce67c, and remained also after upgrading to the latest commit @7c2e15e. Downgrading back to @75ce54b resolves the issue.

RJMSTM, 2023-12-04:
Hello @olliw42 ,

Why you said the problem appeared from the commit **@74ce67c** and this is a commit for RCC IP??
[image]

Thanks,
Rania

olliw42, 2023-12-04:
thx for looking at it

> Why you said the problem appeared from the commit @74ce67c and this is a commit for RCC IP??

The missunderstanding here is that I didn't said what you say I said. I did not say that it appeared from commit @74ce67c. The info I have given implies that the issue occured anywhere from @75ce54b to @74ce67c.

RJMSTM, 2023-12-05:
Hello @olliw42 ,

 I have test an example I2C_TwoBoards_ComDMA with commit @75ce54b to @74ce67c and example works fine,
[image]

[image]

`HAL_I2C_Master_Transmit_DMA `and `HAL_I2C_Slave_Transmit_DMA `API return ok. 

 Please can you give me details in which API your check the problem.

Many Thanks,
Rania

RJMSTM, 2023-12-29:
Hello @olliw42,

We were unable to reproduce your issue, I2c driver work fine in DMA mode. 
However, the point you reported is not directly related to the firmware published in this repository and we don't treat aspect related to user application in our GitHub repositories. They are rather treated at
 [ST Community](https://community.st.com/s/topic/0TO0X000000BSqSWAW/stm32-mcus) level.

Now, as this issue is not directly related to some software component published within this repository (CMSIS, HAL, BSP, etc.) but rather to a user application, please allow me to close it.

Thank you for your comprehension, sorry we will not be able to help you more, we are looking forward to reading from you again.

With regards,
Rania

closed 2023-12-29 by RJMSTM