STMicroelectronics/stm32g4xx-hal-driver #14: HAL_FMAC_FilterStop does not terminate DMA transfer

hobby-pub, 2024-03-24:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus).

**Describe the set-up**
 * Custom STM32G431 board.
 * STM32G4xx HAL Driver 1.5.2.

**Describe the bug**
A wrong behavior of subsequent HAL_FMAC_FilterStart() -> HAL_FMAC_FilterStop() -> HAL_FMAC_FilterStart().

**How To Reproduce**
1. FMAC is configured as FIR with output buffer access mode DMA.
3. The software periodically starts and stops FMAC, by calling HAL_FMAC_FilterStart() / HAL_FMAC_FilterStop().
4. The second call of HAL_FMAC_FilterStart() fails with DMA related HAL error.

**Additional context**
Follow the issue debugging, DMA transfer starts from within HAL_FMAC_FilterStart().
However, complementary DMA stop is missing in HAL_FMAC_FilterStop().

TOUNSTM, 2024-04-02:
ST Internal Reference: 177976

TOUNSTM, 2024-10-16:
Fixed in d69997c6dcdc2d672afae31fee03713fb033c4a8

closed 2024-10-16 by TOUNSTM