STMicroelectronics/stm32g4xx-hal-driver #15: HAL driver always asserts when full assert is activated and i2c speed is set to Fast Mode Plus

rokonrad, 2024-04-12:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * Custom STM32G491 board.
 * STM32CubeIDE 1.15.0 and STM32G4xx HAL Driver 1.2.3

**Describe the bug**
When I2C Speed is set to Fast Mode Plus AND Full Asserts are enabled the Driver always asserts.

**How To Reproduce**
1. Open ioc file

2. Activate checkbox "Enable Full Assert" under Project Manager -> Code Generator -> HAL Settings

3. Set I2C3 Speed Mode to "Fast Mode Plus" under "Pinout & Configuration" -> Connectivity -> I2C3 -> Parameter Settings -> Timing Configuration

4. Save ioc file and let generate code

5. Add a "while (1) {}" in the User Code section into the function assert_failed() in main.c

**Additional context**
The code generator adds the following line to MX_I2C3_Init() in i2c.c:
`__HAL_SYSCFG_FASTMODEPLUS_ENABLE(I2C_FASTMODEPLUS_I2C3);`

This macro is in stm32g4xx_hal.h and includes an assert that leads to:
`assert_param(IS_SYSCFG_FASTMODEPLUS(I2C_FASTMODEPLUS_I2C3))`

The problem is, that this macro checks for
`I2C_FASTMODEPLUS_PB6, I2C_FASTMODEPLUS_PB7, I2C_FASTMODEPLUS_PB8, I2C_FASTMODEPLUS_PB9`
but misses
`I2C_FASTMODEPLUS_I2C1, I2C_FASTMODEPLUS_I2C2, I2C_FASTMODEPLUS_I2C3, I2C_FASTMODEPLUS_I2C4`

A possible fix could be to change both macros __HAL_SYSCFG_FASTMODEPLUS_ENABLE and __HAL_SYSCFG_FASTMODEPLUS_DISABLE to use different asserts:
`assert_param(IS_SYSCFG_FASTMODEPLUS((__FASTMODEPLUS__))` -> `assert_param(IS_I2C_FASTMODEPLUS((__FASTMODEPLUS__))`

KRASTM, 2024-04-17:
ST Internal Reference: 179240

KRASTM, 2024-04-24:
hello @rokonrad,

The issue you reported has been logged into an internal bug tracker it will be managed by the development team.

As the issue is not related to the software published within this repository, rather to the STM32CubeMX, no fix is expected to be published here. Hence, please allow me to close it. Thank you for your comprehension.

With regards,

closed 2024-04-24 by KRASTM

KRASTM, 2024-04-24:
Hello @rokonrad,

It has already been forwarded to the Cube mx team.

With regards.