STMicroelectronics/stm32g4xx-hal-driver #17: [HAL][SPI] State stuck in HAL_SPI_STATE_BUSY_RX when invalid parameters are passed to Rx functions

raqnarok, 2024-07-01:
**Description:**
When using full-duplex SPI in master mode, there is an issue with the state management in the Rx functions. The parameters of the Rx function are checked within the TxRx functions. However, the state `HAL_SPI_STATE_BUSY_RX` is set within the Rx function itself. This leads to a scenario where if the Rx function is called with invalid parameters, the TxRx function returns an error, leaving the SPI state stuck in `HAL_SPI_STATE_BUSY_RX`.

**Steps to Reproduce:**
1. Set up SPI in full-duplex master mode.
2. Call the Rx function with invalid parameters.
3. Observe that the TxRx function returns an error.
4. Notice that the SPI state remains stuck in `HAL_SPI_STATE_BUSY_RX`.

**Expected Behavior:**
The SPI state should not be stuck in `HAL_SPI_STATE_BUSY_RX` when invalid parameters are passed to the Rx functions. Proper parameter validation should occur before changing the state.

**Proposed Solution:**
Move the parameter validation to occur before setting the state to `HAL_SPI_STATE_BUSY_RX` in the Rx functions. This ensures that the state is only set if the parameters are valid.

KRASTM, 2024-07-04:
Hello @raqnarok,

Thank you for the report.
The issue already detected on other series, and Our team will fix it.

With regards,

KRASTM, 2024-07-04:
ST Internal Reference: 185358

PR #16 (2024-07-30): [HAL][SPI] Fix state change issue in Rx functions by validating parameters first
[HAL][SPI] Fix state change issue in Rx functions by validating parameters first

When using full-duplex SPI in master mode, the parameters of the Rx function are checked in the TxRx functions. However, the HAL_SPI_STATE_BUSY_RX state is set in the Rx function. Therefore, if the Rx function is called with invalid parameters, the TxRx function returns an error, leaving the SPI state stuck in HAL_SPI_STATE_BUSY_RX.

Fixes #17