STMicroelectronics/stm32g4xx-hal-driver #20: TEMPSENSOR_CAL2_TEMP incorrectly specified as 110 instead of 130 for STM32G431

jacobq, 2024-09-24:
In `stm32g4xx_ll_adc.h` `TEMPSENSOR_CAL2_TEMP` is set to 110 despite the datasheet saying that it should be 130 (degrees Celsius). I thought this was fixed in v1.5.0, but it appears to be wrong still both in "STM32Cube FW_G4 V1.6.0" and the latest master version here:
https://github.com/STMicroelectronics/stm32g4xx_hal_driver/blob/3d08880e8d7f96454675d48888a3a207388268b0/Inc/stm32g4xx_ll_adc.h#L386

From STM32G431 datasheet [DS12589 Rev 6](https://www.st.com/resource/en/datasheet/stm32g431cb.pdf):
[image]

See https://community.st.com/t5/stm32-mcus-embedded-software/stm32g431-ts-cal2-temperature/td-p/138636

RJMSTM, 2024-09-27:
ST Internal Reference: 192342

RJMSTM, 2024-09-30:
Hello @jacobq,

Value in TEMPSENSOR_CAL2_TEMP does not match the G431 datasheet.TS_CAL2 Temperature sensor raw data acquired value at 130 °C, VDDA = 3.3 V

I have reviewed the other datasheets for G441/G4A1xE/G491/G483/G473/G484/G474. All these part numbers have a TEMPSENSOR_CAL2_TEMP value equal to 130°C.

Regards,

jacobq, 2024-09-30:
> ... VDDA = 3.3 V

Aren't all of these temperature measurements taken with VDDA = VREF+ = **3.0V**?

closed 2024-10-14 by RJMSTM

jannikblf2, 2024-10-18:
@TOUNSTM

With release v1.2.5 (commit d69997c) TEMPSENSOR_CAL2_TEMP is again 110 degrees.