STMicroelectronics/stm32g4xx-hal-driver #24: I2C Slave

bitwisesys, 2025-02-12:
Which structure member contains the number of bytes transferred at the end of a transaction?  
->XferCount does not function in this manner.

KRASTM, 2025-02-14:
Hello @bitwisesys, 

If I understand you correctly, you mean:
`uint16_t                   XferSize;       /*!< I2C transfer size                         */`, in the __I2C_HandleTypeDef struct.

Otherwise, could you be a little more specific?

With regards,

bitwisesys, 2025-02-15:
Hi,

If I call a TX function and for some reason the receiver sends a NAK before everything is read, then TxComplete happens.  How do I know how many characters are actually read?

Thanks,
Blake

KRASTM, 2025-02-18:
Hello @bitwisesys 

I have checked with our team, but unfortunately there are no such parameters for the transferred data in case of NACK. I suggest adding a function to manage this calculation.

Please, allow me to close this issue, thank you in advance for your comprehension.

With regards,

closed 2025-02-20 by KRASTM